SynDaverCO/symple-slicer #177: Disabled settings custom edits

synsuiter, 2022-09-02:
**Is your feature request related to a problem? Please describe.**
When I customize and then and then save a profile with the support enabled turned OFF, AFTER I have edited the settings relating to supports, related setting edits are not saved with the toml file. So when I import it for use, it will have the old settings for support when enabled through the UI. 

**Describe the solution you'd like**
Any edited setting will be saved in the profile regardless of activation. 

**Describe alternatives you've considered**
None.

marciot, 2022-09-02:
Have you checked the TOML file with a text editor to see if the settings are there? I am curious whether it is an issue with the saving, or with the loading.

synsuiter, 2022-09-30:
Due to the inconsistency reproducing this error I am going to close it. We can reopen if it persists.

closed 2022-09-30 by synsuiter