SynDaverCO/symple-slicer #178: NG2 Slicer Settings

synsuiter, 2022-09-23:
**Describe the bug**
Some setting options that are not linked to one another, can be unique from one another, are displaying in the UI as having only one space instead of showing the 2 spaces for each nozzle. 

**Details**
 - SympleSlicer Version: BETA 1.1.1
 - Browser: Google Chrome

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Symple Slicer BETA
2. Go to advanced settings and set your settings view to EXPERT or ALL
3. Apply the NG2 Machine and any material options
4. View the setting options that should have 2 spaces for info
          - EXAMPLE: Under "Resolution", we should be able to assign width of the line to any feature unique to the tool head. If we had a machine with 2 tool heads, one with a 0.4mm nozzle and one with a 0.8mm nozzle. 
          - Line width, wall line width, outer wall line width, inner wall line width, and top/bottom line width. 

**Expected behavior**
There should be 2 boxes to input a value. 

Another Idea is that we have 2 spots for every option but the value is liked for some so they just auto change to match, just an idea. 

**I am only mentioning the **Resolution** settings here but there are more examples in other sections.**

[image]

marciot, 2022-09-23:
@synsuiter: If you look in fdmprinter.def.json you will see that "Line Width" (i.e. line_width) does not have the attribute "settable_per_extruder", thus simple slicer will not consider it settable per extruder.

However, weirdly, this presents a paradox, as "line_width" is based on the value "machine_nozzle_size", which is "settable_per_extruder". Can you maybe investigate how Cura handles the paradox? Does it automatically assume "settable_per_extruder" when a value depends on another which is "settable_per_extruder"? If so, we could potentially add the same assumption to Symple Slicer.

synsuiter, 2022-09-23:
I believe it does because the Cura Ultimaker has those, and more, settings as individual per extruder in their UI. 

Another thing to consider, the nozzle diameter might define the width but the width can then be set independent of the nozzle diameter as well. 
[image]

synsuiter, 2022-10-26:
I was able to find that the settings in Cura Ulti that are showing as settable_per_extruder, actually contain "settable_per_mesh": true.

If the extruders are assigned to a different mesh then they have to technically be able to set per extruder since we don't have the ability to assign slice features to a mesh, we are having the settings all go through extruder assignment. 

I am unsure if this is exactly how they are allowing more individual edits but it could be worth looking into.

==================================================================================

I also discovered that the Support settings are ALL supposed to be NOT settable per extruder since you must assign a shared extruder for all supports generated, but in our UI, they are ALL showing as being able to edit independently. I double checked and these settings DO have "settable_per_extruder": false, AND "settable_per_mesh": false, under each of them in the FDM file.  

Let me know your thoughts on this, I am going to keep digging.

synsuiter, 2022-10-26:
**Line width options also have these:
                    "value": "machine_nozzle_size",
                    "settable_per_mesh": true,

synsuiter, 2022-10-26:
@marciot , I couldn't find this online anywhere but this is what I think we were missing.

Settable_per_exrtuder = Can assign a specific extruder to this feature regardless of mesh assignment. 

Settable_per_mesh = Can be assigned unique values per mesh, which can assigned to different extruders. 

So it appears that “settable per extruder” actually refers back to having a specific extruder assignment, which actually means the feature, regardless of mesh, can be assigned to a single extruder. 

So like the Build Plate adhesion stuff IS settable per extruder, but only because it is ASSIGNED a specific extruder. The Support settings are the same, only one extruder can be used when generating support, so it is settable per extruder. But speed, for example, is settable per mesh, which means it can be unique to the single extruder assigned to the mesh. 

BOTH = The settings that share the “settable per mesh” AND “settable per extruder” can be both individual different settings, OR it can be assigned to a specific extruder and shared. These will require some link to the Extruder assignment setting options.

marciot, 2022-10-31:
@synsuiter: I see. This makes sense. 

Symple Slicer does not support per mesh settings and adding that option would require a pretty major overhaul (in the order of a several months), so it seems like one solution for now is to show two separate values for any setting which is either "settable_per_extruder" or "settable_per_mesh". In the case in which Cura allows a setting to be assigned by mesh, Symple Slicer would only allow the user to configure that by extruder and the mesh would take on the value corresponding to the extruder it is assigned. So basically, in Symple Slicer, "settable_per_mesh" would be synonymous with "settable_per_extruder".

Would that work, at least in the interim?

synsuiter, 2022-11-10:
Where are we on this? I see there are still settings that have "settable_per_mesh" but are not visible for the 2nd Extruder. Was there a file I was supposed to update and send you?

marciot, 2022-11-11:
@synsuiter: What settings are missing?

synsuiter, 2022-11-11:
I have confirmed the following settings are: "settable_per_mesh = true"

[image]
In the Resolution section, the following settings are missing for Extruder 2: 
Wall Line Width
Outer wall line width
Inner wall(s) line width
Top/bottom Line Width
Infill Line Width (when not overridden)

[image]
In the Walls section, the following settings are missing for Extruder 2, when NOT OVERRIDEN: 
Wall thickness
Wall line count
Outer wall Wipe distance
Outer wall inset
Alternate Extra Wall
Compensate Wall Overlaps
Compensate Outer Wall Overlaps
Compensate Inner Wall Overlaps
Fill Gaps Between Walls
Filter out Tiny Gaps
Print Thin Walls
Horizontal Expansion
Initial Layer Horizontal Expansion
Hole Horizontal Expansion
Z seam Alignment (and those related to alternate options)
Seam Corner Preference (and those related to alternate options)

[image]
In the Top/Bottom section, the following settings are missing for Extruder 2, when NOT OVERRIDEN:
Top Surface Skin Layers
Top/Bottom Thickness
Top Thickness
Top Layers
Bottom Thickness
Bottom Layers
Initial Bottom Layers
Top/Bottom Pattern
Bottom pattern Initial Layer
No Skin in Z gaps
Extra skin wall count
Enable Ironing
Skin Overlap Percentage
Skin Overlap
Skin removal Width
Top Skin removal Width
Bottom Skin removal Width
Skin Expand Distance 
Top Skin Expand Distance 
Bottom Skin Expand Distance 
Maximum Skin angle for expansion
Minimum Skin width for expansion 

[image]
In the Infill section, the following settings are missing from Extruder 2, when NOT OVERRIDEN: 
Infill Density 
Infill Line Distance 
Infill Pattern
Connect Infill Lines
Randomize Infill Start
Infill Multiplier
Extra Infill wall Count
Infill Overlap Percentage
Infill wipe Distance
Infill Layer Thickness
Gradual Infill Steps
Minimum Infill Area
Infill Support
Skin Edge Support Thickness
Skin Edge Support Layers

[image]
In the Material Section, the following settings are missing from extruder 2:
Wall Flow
Outer Wall Flows
Inner Walls(s) Flow
Top/Bottom Flow
Infill Flow

[image]
In the Speed section, the following settings are missing from Extruder 2, if NOT OVERRIDEN in a previous section:
Infill Speed
Outer Wall Speed
Inner wall speed
Top/Bottom Speed
Infill Acceleration
Outer Wall Acceleration
Inner Wall Acceleration
Top/Bottom Acceleration
Infill Jerk
Outer Wall Jerk
Inner Wall Jerk
Top/Bottom Jerk

There may be more that I haven't listed because they are not visible until something else is active, like Acceleration Control, etc.
Some of these settings need to only be visible if the extruder assignment isn't specified (aka Not Overridden), The Support settings look they they are all working well but those are the only ones that MUST be assigned so showing only one column for those works. In the images above, I added the models side by side so you can see that if not specified, they should be able to have their own values for the above settings.

marciot, 2022-11-11:
@synsuiter: I see. I looks like some settings are limited to a particular extruder only when certain conditions are met, and I did not account for this. So I will need to rework things again to account for this additional complexity.

synsuiter, 2022-11-11:
okydoky, makes sense.

synsuiter, 2022-11-11:
Looks great, closing ticket!

closed 2022-11-11 by synsuiter