SynDaverCO/symple-slicer #179: Large Model Slicing Issues

synsuiter, 2022-09-29:
**Describe the bug**
While testing a larger build volume we found some limits in slicing when creating a custom bed size. 
A 20mm cube model succeeds slicing at up to 974mm in size, but fails at 974.1 or larger in size. 

Debug Console error message: "Warning: Unable to strip Cura header" which is an intentional error message implemented on line 95 of src-app/lib/slicing-engines/CuraEngine/SlicerWorker.js. Possibly, whatever is causing this is happening prior to that and this is just catching something being wrong.

This is the most straight forward example, we did have failures with other models that were more complex, like the scaled up "squid attack". 

**Details**
 - SympleSlicer Version: USER Release version - 1.1.0
 - Browser : Chrome

**To Reproduce**
Steps to reproduce the behavior:
1. Load User SS
2. Apply an AXI 2 profile
3. Edit machine settings for the build area to equal a square meter
4. import 20mm cube model
5. increase model size and slice

**Expected behavior**
SS should be able to 

**Additional context**
Profile Changes made:
AXI 2 Profile with machine size edited to be 1000 X1000 X1000mm
Build Plate adhesion type was changed to "none".
Relevant models and gcodes have been attached.
[974p1mm_cube.txt](https://github.com/SynDaverCO/symple-slicer/files/9679526/974p1mm_cube.txt)
[20mm_cube-stl.txt](https://github.com/SynDaverCO/symple-slicer/files/9679528/20mm_cube-stl.txt)

Unable to attach 974mm.gco due to large file size. Please see Teams.

marciot, 2022-09-30:
@synsuiter: Have you tested this under Ultimaker Cura? It may be a limitation on the slicer engine, rather than Symple Slicer.

synsuiter, 2022-09-30:
Samantha got this error:

Cannot enlarge memory, asked to go up to 2148188160 bytes, but the limit is 2147483648 bytes!
Cannot enlarge memory, asked to go up to 2148192256 bytes, but the limit is 2147483648 bytes!
Aborted(native code called abort())
filename:  https://syndaverco.github.io/slicer-beta/lib/slicing-engines/CuraEngine/CuraEngine.js  lineno:  1  error:  Uncaught RuntimeError: Aborted(native code called abort())

marciot, 2022-09-30:
Added an alert box to let the user know that the slicing engine has failed and direct them to check the slicing logs for details. Not really a fix, but something to make it easier to troubleshoot.

synsuiter, 2022-09-30:
Repeated on current BETA version, cube at 980 cubed

Starting slice...
Slicing model...
slice of mesh took 0.013 seconds
slice make polygons took 0.019 seconds
Progress: slice accomplished in 0.032s
Starting layerparts...
Progress: layerparts accomplished in 0.022s
Starting inset+skin...
Layer count: 3920
Progress: inset+skin accomplished in 0.587s
Starting support...
Processing platform adhesion
Progress: support accomplished in 0.238s
Starting export...
Cannot enlarge memory, asked to go up to 2522345472 bytes, but the limit is 2147483648 bytes!
Cannot enlarge memory, asked to go up to 2522349568 bytes, but the limit is 2147483648 bytes!
Aborted(native code called abort())
filename:  https://syndaverco.github.io/slicer-beta/lib/slicing-engines/CuraEngine/CuraEngine.js  lineno:  1  error:  Uncaught RuntimeError: Aborted(native code called abort())

synsuiter, 2022-09-30:
Reduced Cube to 973 but still got failure. 

Starting slice...
Slicing model...
slice of mesh took 0.014 seconds
slice make polygons took 0.021 seconds
Progress: slice accomplished in 0.036s
Starting layerparts...
Progress: layerparts accomplished in 0.019s
Starting inset+skin...
Layer count: 3892
Progress: inset+skin accomplished in 0.568s
Starting support...
Processing platform adhesion
Progress: support accomplished in 0.240s
Starting export...
Cannot enlarge memory, asked to go up to 2516889600 bytes, but the limit is 2147483648 bytes!
Cannot enlarge memory, asked to go up to 2516893696 bytes, but the limit is 2147483648 bytes!
Aborted(native code called abort())
filename:  https://syndaverco.github.io/slicer-beta/lib/slicing-engines/CuraEngine/CuraEngine.js  lineno:  1  error:  Uncaught RuntimeError: Aborted(native code called abort())

marciot, 2022-09-30:
@synsuiter, @SynWillinger: After some investigation I found out by default the web environment only supports 2GB of memory (this is exactly 2147483648 bytes). There is an option to increase that limit to 4GB at compile time, but as described in this thread, it breaks compatibility with certain browsers, although the thread does not say which:

https://github.com/emscripten-core/emscripten/issues/16197

Hence, at the moment, we risk breaking Symple Slicer for some users if we compile a version of Symple Slicer with a 4GB memory limit. I suggest we try seeing if there is a way to set the slice parameters so less memory is used, before we commit to making a possibly incompatible change for all users.

marciot, 2022-09-30:
I got a different error when slicing `dragon_nice_1.stl` at 500%

```
filename:  http://localhost:8000/lib/slicing-engines/CuraEngine/CuraEngine.js  lineno:  1  error:  Uncaught RangeError: Array buffer allocation failed
```

This seems to expose the same problem under two different circumstances:

1. The cases in which the slicing engine needs more than 2GB of RAM to slice the model
2. The case in which the resulting GCODE file is too large, which is the later error.

marciot, 2022-10-03:
@symsuiter, @synwillinger: I'm going to have to declare this one a "won't fix", or more accurately, "can't fix". Currently, web browsers only support a 32-bit environment (with the corresponding 4GB limit) and after some experimentation it's become evident to me that slicing for a large format printer will require a 64-bit environment. There are also speed issues to contend with. Attempting to slice a large, complex model in the web browser took several hours -- doing so in native Ultimaker Cura took a few dozen minutes. Huge difference.

I think the only practical route is to either use Ultimaker Cura (or another slicer), or to develop a desktop version of Symple Slicer using Electron that calls a native, 64-bit command-line CuraEngine on the back end. By doing so, we will have to deal with many issues we have been avoiding by having a web-based slicer, such as having to compile for different platforms, having to deal with code signing, having to provide downloadable updates, etc.

marciot, 2022-12-02:
Adding information to this ticket for future reference. Symple Slicer Desktop runs into a different limit, a limitation of 512MB for post-processing the GCODE. Slices produces the following error, which is due to a limitation is node.js:

```
SlicerInterface.js:261 Error: Cannot create a string longer than 0x1fffffe8 characters
    at Object.slice (node:buffer:592)
    at Buffer.toString (node:buffer:810)
    at readFileHandle (node:internal/fs/promises:371)
    at async ChildProcess.onExit (SlicerInterface.js:256)
```