SynDaverCO/symple-slicer #184: Visual error in Navigation Cube

synsuiter, 2022-10-14:
**Describe the bug**
The "back" and "front" walls of the cube become invisible after selecting the "back" option. 

**Details**
 - SympleSlicer Version: BETA 1.2.0.
 - Browser : google Chrome

**To Reproduce**
Steps to reproduce the behavior:
1. Open SS and navigate with the cube right click menu options. 

**Expected behavior**
walls should remain visible. 

**Screenshots**
[image]

marciot, 2022-10-14:
This one might be a web browser rendering bug and there may be nothing I can do about it. It's very peculiar.

synsuiter, 2022-10-20:
I cant seem to replicate it now so I am going to close this out. we can reopen if it becomes in issue later.

closed 2022-10-20 by synsuiter