SynDaverCO/symple-slicer #185: Add previously absent Dual Settings

synsuiter, 2022-10-24:
**Is your feature request related to a problem? Please describe.**
Need to be able to edit these Dual specific Setting options.

**Describe the solution you'd like**
Here is the list of settings

wall_extruder_nr
wall_0_extruder_nr
wall_x_extruder_nr
roofing_extruder_nr
top_bottom_extruder_nr
infill_extruder_nr
support_extruder_nr
support_infill_extruder_nr
support_extruder_nr_layer_0
support_interface_extruder_nr
support_roof_extruder_nr
support_bottom_extruder_nr
adhesion_extruder_nr
skirt_brim_extruder_nr
raft_base_extruder_nr
raft_interface_extruder_nr
raft_surface_extruder_nr
prime_tower_enable
prime_tower_size
prime_tower_min_volume
prime_tower_position_x
prime_tower_position_y
prime_tower_wipe_enabled
prime_tower_brim_enable
ooze_shield_enabled
ooze_shield_angle
ooze_shield_dist
switch_extruder_retraction_amount
switch_extruder_retraction_speeds
switch_extruder_retraction_speed
switch_extruder_prime_speed
switch_extruder_extra_prime_amount

**Describe alternatives you've considered**
None

marciot, 2022-10-24:
Please modify the following file and send me an updated copy:

https://github.com/SynDaverCO/symple-slicer/blob/master/src-app/js/SlicerSettings.js

synsuiter, 2022-10-24:
file updated

synsuiter, 2022-11-11:
Dual settings are now visible and usable, If there are future issues with Dual settings I will open a new ticket.

closed 2022-11-11 by synsuiter