SynDaverCO/symple-slicer #187: Extra Start Gcode

synsuiter, 2022-11-01:
**Describe the bug**
There is extra gcode being generated at the very beginning of the file. See examples below. 
If this code is deleted manually, the gcode is usable but if left in, the print will not begin correctly. 

**Details**
 - SympleSlicer Version: 1.2.0
 - Browser: Chrome

**To Reproduce**
Steps to reproduce the behavior:
Slice any model with any machine or material, open the gcode file once saved. 

**Expected behavior**
There should not be this extra text at the start of a print. 

Example of extra text
;FLAVOR:Marlin
;TIME:6666
;Filament used: 0m
;Layer height: 0.5
;MINX:2.14748e+06
;MINY:2.14748e+06
;MINZ:2.14748e+06
;MAXX:-2.14748e+06
;MAXY:-2.14748e+06
;MAXZ:-2.14748e+06

;Generated with Cura_SteamEngine master
T0
M140 S80
M105
M190 S80
M104 S235
M105
M109 S235
M82 ;absolute extrusion mode
;This G-Code has been generated specifically for NG2 pre-heatbed...

synsuiter, 2022-11-04:
Upon further investigation this seems like a new Engine thing, not a Bug. Can we just remove that from the generated gcode? or have it not add it in the 1st place? It is doing it on Cura Ultimaker also. 

Cura Ulti Version # = 

text in Ulti generated gcode: 5.0.0.

;Generated with Cura_SteamEngine 5.0.0
T0
M140 S60
M105
M190 S60
M82 ;absolute extrusion mode

marciot, 2022-11-11:
@synsuiter: Can this be closed?

closed 2022-11-11 by synsuiter