SynDaverCO/symple-slicer #188: Centering Error for larger models

synsuiter, 2022-11-01:
**Describe the bug**
When centering larger models, it places then on the top left, instead of in the middle.

Seems t affect models that come close to exceeding the AXI2 Bed limits. Not sure if that is relevant. 

**Details**
 - SympleSlicer Version: 1.2.0
 - Browser: Chrome

**To Reproduce**
Steps to reproduce the behavior:
1. Apply any printer and material profile
2. Load in a large model, I have attached an example model.
3. Try to center the model.

**Expected behavior**
Center model to build plate bed

**Screenshots**
This is how it "centers" now:
[image]

This is the model shrunk down and centered, it centers correctly. 
[image]

These are the size dimensions of the 1st larger model, it SHOULD have been able to center and print, it was not larger than the build plate, I excluded brim and skirt from the settings in case the UI was picking up on that causing the print to not fit. 
[image]

marciot, 2022-11-07:
@synsuiter: Can you attach the STL to this ticket?

synsuiter, 2022-11-07:
My apologies, I meant to and completely forgot. 
[fullbed_wavytest-superfine.txt](https://github.com/SynDaverCO/symple-slicer/files/9954087/fullbed_wavytest-superfine.txt)

Change to .stl

synsuiter, 2022-12-09:
fixed

closed 2022-12-09 by synsuiter