SynDaverCO/symple-slicer #190: Unable to generate Gcode

synsuiter, 2022-11-07:
**Describe the bug**
Upon slicing a normal model with a normally applied Machine and Material profile, BUT THEN enabling the SUPPORTS the slicer generated 0 lines of code. 

**Details**
 - SympleSlicer Version: 1.2.0
 - Browser: Google Chrome, up to date. 
         - Browser history and data wiped. Error confirmed on multiple computers.

**To Reproduce**
Steps to reproduce the behavior:
1. Apply a profile
2. Load a model
3. Activate supports
4. slice and see redult

**Screenshots** Pictures Below 

[image]
[image]

synsuiter, 2022-11-07:
[image]
[image]

marciot, 2022-11-07:
@synsuiter: This appears to be a problem with CuraEngine requiring "support_enabled" to be presented in the command line before certain other settings. I have created a ticket upstream.

https://github.com/Ultimaker/CuraEngine/issues/1763

We can work around this on our end, although there is a posibility other settings may have similar requirements.

marciot, 2022-11-11:
@synsuiter: Can this be closed?

synsuiter, 2022-11-11:
Indeed, Issue has been resolved. Gracias

closed 2022-11-11 by synsuiter