SynDaverCO/symple-slicer #191: Dual Profile export/import

synsuiter, 2022-11-11:
**Describe the bug**
I exported a dual profile and got this at the very bottom:

[settings.extruder_0]
cool_fan_full_at_height = **null**                            # mm
cool_fan_full_layer = 15
cool_fan_speed_max = 100                                  # %
cool_fan_speed_min = 100                                  # %
retraction_amount = 5                                     # mm
retraction_speed = 45                                     # mm/s
speed_print = 50                                          # mm/s
speed_wall = 50                                           # mm/s

[settings.extruder_1]
cool_fan_enabled = false
cool_fan_full_at_height = 2                               # mm
cool_fan_speed_max = 50                                   # %
cool_fan_speed_min = 50                                   # %
material_flow_layer_0 = 110                               # %
retraction_amount = 6                                     # mm
retraction_speed = 60                                     # mm/s
speed_print = 40                                          # mm/s
speed_wall = 40                                           # mm/s

But I cant import and use it cuz the "cool_fan_full_at_height = null" isnt recognizable by the slicer. It is also possible it is having an issue with repeated settings, any repeats will cause the profile import to not work. 

**Details**
 - SympleSlicer Version: 1.2.0
 - Browser : google chrome

**To Reproduce**
Steps to reproduce the behavior:
1. apply dual machine
2. export dual profile 
3. import same profile
4. see error (image below) 

**Expected behavior**
Unsure what behavior is causing the error

**Screenshots**
[image]

synsuiter, 2022-11-11:
I changed the "null" to a numerical value and the profile WAS able to import without issue.

synsuiter, 2022-11-11:
@marciot , Can we make it so if a value is "null" or blank when exported, that it wont allow the profile export or it comments that line out or removes that line entirely?

marciot, 2022-11-11:
@synsuiter: This is a symptom of some other problem, as there should not be null values in the backend. I will need to investigate.

synsuiter, 2022-11-18:
I have confirmed that if a setting is just deleted, no value to replace it, the profile will put "null" as the setting Value when exported. 

another thing to note from this, the slicer is allowing for slicing when setting values are left blank.

Could we have it automatically add the applied profile default value if left empty?

marciot, 2022-11-21:
@synsuiter: For consistency with ticket #194, I now make it so if the user "erases" a value in the UI, it will restore to the Cura default and a blue value will become white. This essentially allows the user to remove values that were either manually entered or entered from the profile.

I know this is not what you asked for, but it is the best I can do right now.

synsuiter, 2022-11-21:
that looks great to me, I think this at least helps out, I will close this ticket and we can make a new one for any future advancements to this feature. 

Just a note for later: the Darkness Mode doesn't really show the color difference but idk if that is a concern at this moment.

closed 2022-11-21 by synsuiter

marciot, 2022-11-21:
@synsuiter: In Darkness, there is a color difference. Gray vs light blue. It's not all that pronounced though.