SynDaverCO/symple-slicer #192: Print Time Est for Dual Prints

synsuiter, 2022-11-11:
**Describe the bug**
The Time est for Dual prints are WAAAAAY over what the actual print time is, by a factor of days.

A print repeated almost exactly in the Ultimaker version comes out to 4 hours where Symple slicer has it at 71 hours. they actually print at the same speed and don't have a significant difference when actually printed and compared. 

**To Be Noted** SS is also adding long travel moves (to X0 and Y0) in between each nozzle switch, while this does add extra time to the print it does not add several days of time, and we have removed those travel moves manually and observed it print, behavior is identical.

**Details**
 - SympleSlicer Version: 1.2.0
 - Browser : Google Chrome

**To Reproduce**
Steps to reproduce the behavior:
1. Make a dual slice and observe the time est

**Screenshots**
Example here of 2 Cubes being printed side by side in two different materials. 

[image]
[image]
[image]

marciot, 2022-11-11:
@synsuiter: Please look at the GCODE and see what time is stated in the header. This is what Symple Slicer uses.

synsuiter, 2022-11-11:
The one that SS est at 71 hours and ULTI had 4 hours had this:

;FLAVOR:Marlin
;TIME:259066
;Filament used: 132.644m, 24.0319m
;Layer height: 0.5
;MINX:0
;MINY:0
;MINZ:0.5
;MAXX:550.877
;MAXY:624.433
;MAXZ:246
M82 ;absolute extrusion mode
;This G-Code has been generated specifically for NG2

And the Cube one, claiming 27 hours, had this: 
;FLAVOR:Marlin
;TIME:99281
;Filament used: 21.2532m, 20.9991m
;Layer height: 0.5
;MINX:0
;MINY:0
;MINZ:0.5
;MAXX:551.198
;MAXY:570.808
;MAXZ:51.5
M82 ;absolute extrusion mode
;This G-Code has been generated specifically for NG2

synsuiter, 2022-11-11:
Here are those gcodes
[20mm_cubes_time-est - Copy.txt](https://github.com/SynDaverCO/symple-slicer/files/9993130/20mm_cubes_time-est.-.Copy.txt)
[cadet-head-right_NG2_PLA+PVA_11-10-22 - Copy.txt](https://github.com/SynDaverCO/symple-slicer/files/9993131/cadet-head-right_NG2_PLA%2BPVA_11-10-22.-.Copy.txt)

marciot, 2022-11-11:
@synsuiter: 259066 seconds works out to 71 hours, so Symple Slicer is showing the correct time, at least as far as the engine is reporting. You may want to make sure you are comparing times against Ultimaker Cura 4.12.1, as that is what is currently in Symple Slicer.

synsuiter, 2022-11-11:
Here are the exact comparisons, same settings and Engine, 4.12.1.

The Symple Slicer Est is 121 Hours and 47 Minutes
The Cura Ultimaker Est is 6 Hours and 19 Minutes

[image]

marciot, 2022-11-11:
@symsuiter: Are the output GCODE exactly the same? My guess is they aren't, because the profiles are not exactly the same. The only way to really compare the two adequately is to make sure all the slicer settings are exactly the same.

synsuiter, 2022-11-11:
I did, I set them and went over every single setting side by side

synsuiter, 2022-11-11:
here are the two gcodes
[cadet_head_Ulti-time-est.txt](https://github.com/SynDaverCO/symple-slicer/files/9993368/cadet_head_Ulti-time-est.txt)
[cadet_head_ss-time-est.txt](https://github.com/SynDaverCO/symple-slicer/files/9993369/cadet_head_ss-time-est.txt)

synsuiter, 2022-11-11:
They are not identical but I set the setting to match so this may go deeper than just the time est.

synsuiter, 2022-11-11:
I am noticing the Symple Slicer version has a repeat of the start gcode part the way down, very odd

synsuiter, 2022-11-18:
This was likely due to the erroneous travel moves ticket #193, closing ticket. Will reopen if new info comes to light.

closed 2022-11-18 by synsuiter