SynDaverCO/symple-slicer #193: Extra Travel Moves

synsuiter, 2022-11-11:
**Describe the bug**
SS is adding extra travel moves to X00 Y00 each time the nozzles switch. 

**Details**
 - SympleSlicer Version: 1.2.0
 - Browser : google chrome

**To Reproduce**
Steps to reproduce the behavior:
1. Create a dual slice with both nozzles being used to print.
2. observe the preview travel lines.

I tried changing the setting "layer_start_x" and "layer_start_y" to the center of the bed (500x500) but that didn't change where it was traveling too. 

**Screenshots**
[image]

marciot, 2022-11-11:
@synsuiter: This is a slice engine issue. I believe I've seen it before and Ultimaker Cura 4.12.1 does the same thing.

marciot, 2022-11-14:
@synsuiter: Please test with "machine_extruder_end_pos_abs" set to "false" and possibly also "machine_extruder_start_pos_abs". I believe this is the setting that is causing the problem.

synsuiter, 2022-11-14:
Will do

synsuiter, 2022-11-16:
This works, I have pushed machine profile edits for the NG2. Will close when applied and confirmed.

marciot, 2022-11-18:
@synsuiter: I pulled, but there are no changes to the profiles.

synsuiter, 2022-11-18:
That..... Is because I forgot to Push them. My B. sending your way now.

synsuiter, 2022-11-18:
This is solved, good find sir.

closed 2022-11-18 by synsuiter