SynDaverCO/symple-slicer #194: "Edited" color identifier

synsuiter, 2022-11-11:
**Is your feature request related to a problem? Please describe.**
I am unable to remember/identify the settings I have edited and the ones that are still from the "applied" profile if I am interrupted during a slice. This leaves me to have to reAPPLY a fresh profile and restart my edits. 

**Describe the solution you'd like**
When the user edits a setting, there could be some kind of visual change indicating that that setting is no longer matching the Applied profile.

**Describe alternatives you've considered**
Not sure of any other alternatives at this time.

marciot, 2022-11-21:
@symsuiter: It is not possible to color code settings which differ from the profile, as Symple Slicer does not distinguish whether a change was made by a profile, or later by the user. Think of applying a profile as a short cut for the user manually typing in all those individual values in the UI. What Symple Slicer does do is mark those values as user-entered and makes it so they no longer affected by the formulas when other values change. Because these values are customized from the defaults, it does indeed make sense to highlight them in the UI, so they are now colored blue.

You will notice that when you load a profile now, certain settings will be colored blue. These are the values which are present in the TOML file (and not commented out). The white settings are not present in the TOML file and are computed based on the formulas or Cura defaults.

synsuiter, 2022-11-21:
This works for me, will open new ticket if changed are needed in future.

closed 2022-11-21 by synsuiter