SynDaverCO/symple-slicer #196: Error(?) in TOML export for DUAL

synsuiter, 2022-11-22:
**Describe the bug**
See files attached containing example. There are some settings that need to have different definitions and it is all currently lumped in the machine profile. I was wondering if now is a the time to to tool head profiles that pair to the machines?

I am not entirely sure what the issue is here or how to fix it. 

**Details**
 - SympleSlicer Version: create a Dual Slice and export the profiles
 - Browser : NG2

**To Reproduce**
Steps to reproduce the behavior:
1. Apply the NG2 PLA
2. export
3. see file sections pertaining to tool heads. 

**Expected behavior**
Unsure...

**Screenshots**

**EXAMPLE OF CURA ULTI MACHINE PROFILE**
[syndaverng2.def.txt](https://github.com/SynDaverCO/symple-slicer/files/10070844/syndaverng2.def.txt)

**EXAMPLE OF CURA ULTI TOOLHEAD PROFILES***
[syndaverng2_extruder_0.def.txt](https://github.com/SynDaverCO/symple-slicer/files/10070845/syndaverng2_extruder_0.def.txt)
[syndaverng2_extruder_1.def.txt](https://github.com/SynDaverCO/symple-slicer/files/10070846/syndaverng2_extruder_1.def.txt)

**SYMPLE SLICER FULL MACHINE AND EXTRUDER PROFILE:**
[dual-profile-export.txt](https://github.com/SynDaverCO/symple-slicer/files/10070863/dual-profile-export.txt)

marciot, 2022-11-28:
@synsuiter: Last Monday it looks like I seriously broke things. So basically, loading dual profiles was not working at all. I made some progress cleaning up the mess on Friday, but still have some more to do today. Hopefully I can upload a new version today and we can see if it closes this ticket and some of the others.

synsuiter, 2022-11-28:
The settings that I am most concerned about are listed bellow because they are/ can be different values from Toolhead 0 to Toolhead 1.

"extruder_nr"     (ex. 0 and 1)
"machine_nozzle_size"     (ex. 0.4 and 1.0)
"machine_extruder_cooling_fan_number"     (ex. 0 and 1)
"machine_nozzle_id"      (ex. Hemera0 and Hemera 1)

synsuiter, 2022-11-28:
**to be clear, the extruder settings have the potential to all be different from one anther but the ones listed above are the ones we need to be able to independently assign.

marciot, 2022-11-28:
@synsuiter: You can specify extruder specific settings in a machine profile, or any profile file for that matter (although it would probably not make sense for anything other than machine profiles). Just use the same notation that is used in exported TOML files:

```
[settings]

...

[settings.extruder_0]
extruder_nr = 0
...

[settings.extruder_1]
extruder_nr = 1
...
```

synsuiter, 2022-11-28:
okydoky, got it

synsuiter, 2022-12-09:
our new machine profile format, with the separated extruder definitions, seems to be working well. All exports have everything unique listed.

closed 2022-12-09 by synsuiter