SynDaverCO/symple-slicer #198: "All Cura Settings" not showing any settings

SynWillinger, 2022-12-02:
**Describe the bug**
Selecting "all cura settings" as the slicer settings results in no settings showing up 

**Details**
 - SympleSlicer Version: 1.2.0 Desktop build sent 12-2-22
 - Browser: none, desktop

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Advanced Features in the page drop-down
2. Expand "User Interface" section
3. For "Slicer Settings" select "All Cura Settings"
4. Apply setting and wait for SS to reload
5. Go to "Slice Objets" in the page drop-down
6. See issue

**Expected behavior**
All settings should be available

**Screenshots**
[image]

**Additional context**
Expert settings does work

synsuiter, 2022-12-07:
also in axi machine view.
[image]

synsuiter, 2022-12-09:
Confirmed Still Present. 12/9/22

synsuiter, 2022-12-12:
corrected

closed 2022-12-12 by synsuiter